opensensorhub/osh-js #32: Question:  FFMPEGJS

colek42, 2016-11-23:
Impressive work with the video overlay in Cesium!  I am looking to do the same thing but instead of demuxing the video and sending only NAL units I want to send TS packets proxied from a UDP stream.  Our video actually has sensor data embedded as a data stream and uses the KLV standard.  

How would you add support for a mpeg-ts demuxer in ffmpegjs?

alexrobin, 2016-11-23:
Thanks for your interest.

My fork of the ffmpeg.js repo is [here](https://github.com/sensiasoft/ffmpeg.js). I added a new target called 'ffmpeg-h264' in the main [Makefile](https://github.com/sensiasoft/ffmpeg.js/blob/master/Makefile). You can do the same to add mpeg decoder and mpeg-ts demuxer support. This is all done via FFMPEG [compile flags](https://github.com/sensiasoft/ffmpeg.js/blob/master/Makefile#L299).

Good luck. Please let us know how it goes!

closed 2016-11-23 by alexrobin

colek42, 2016-11-28:
avformat is all I needed.  It worked great!  

my code is at  https://github.com/colek42/streamingDemo.  It looks like you can just send the ```Packet.data []uint8``