opensensorhub/osh-js #33: Add Bower package support

richard3d, 2016-12-29:
Need Bower package support for client-side integration. Bower is intended for client-side packages and is stricter with package management/versioning. This will be mandatory for larger scale projects such as GeoNode/Maploom integration where backend packages are handled by NPM and front-end is handled by Bower. Bower packages typically only include distributable versions of the code as well, versus everything needed for development like NPM.

richard3d, 2017-01-14:
I've migrated most of the local vendor packages to be installed via Bower now. However several packages do not support Bower installation. The list I have so far is Broadway, jQRangeSlider, Tree, and possibly ffMPEG. @mdhsl can you give me a list of urls where you downloaded the source files for the above?

mdhsl, 2017-01-14:
JQRangeSlider is no longer needed anymore, we can just remove it as well as Broadway. FFMpeg is the library which has  been compiled using Emscripten: https://github.com/sensiasoft/ffmpeg.js. For the tree.js: https://github.com/rafaelthca/aimaraJS

alexrobin, 2017-01-16:
I think we still need to keep Broadway as an optional module. It's much
more compact than the FFMPEG JS we created and works fine with some video
feeds.

----------------------------------
Alexandre Robin
Sensia Software, LLC
www.sensiasoftware.com

richard3d, 2017-01-17:
Ok I kept Broadway for the time being and canned JQRangeSlider. For libs that don't include a straight-up distribution (i.e. FFMpeg, tree.js, etc), I employed the concept of a 'vendor-local' directory also. The commit message has all the details.

closed 2017-01-17 by richard3d