opensensorhub/osh-js #84: Add flag to control connection on startup

alexrobin, 2017-08-18:
The DataReceiverController connectAll() method always connects all data sources. In order to save bandwidth and improve client responsiveness, we should be able to control which data sources should be connected on startup. This can be done by adding a boolean flag called "connect" in datasource options, and checking that flag in connectAll(). It can default to true to keep backward compatibility.

mdhsl, 2017-08-18:
There is another possible way to connect datasource. This way would be preferred to the connectAll() method anyway:

`OSH.EventManager.fire(OSH.EventManager.EVENT.DATA+"-"+dataSourceId, {data : data});`

Here are some details about it (DATA para):

[http://opensensorhub.github.io/osh-js/Documentation/dev/technical/](url)

Does this meet your requirement?

alexrobin, 2017-08-18:
I think connectAll() needs to be called anyway to start the sync buffer.
Plus it's nice to be able to configure the data sources and then connect
them all in one call.
The need is similar to the "show" property we have on view, we use it as a
startup default even though views can be opened and closed individually
after that.

---
Alex Robin
+1 302-703-7072
www.sensiasoft.com

commit bce50237 (2017-08-28): Added flag to control if datasources are connected on startup. Fixes #84
closed 2017-08-28 by alexrobin