opensensorhub/osh-js #156: Question: How do you compile ffmpeg-h264.js

MathisTLD, 2020-06-05:
Hello guys I read this article : 
https://opensensorhub.org/2016/09/16/video-decoding-in-osh-js-toolkit/

but couldn't find a trace of your ffmpeg.js fork
I finally found the build file here : 
https://github.com/opensensorhub/osh-js/blob/master/source/osh/ui/view/video/workers/ffmpeg-h264.js

but i definitely would like to know how you compile it  and i would appreciate if you can tell me more about H265 and other codecs you mentioned in the article

thanks for reading, have a great day

alexrobin, 2020-06-05:
Hi there, thanks for your interest. The latest work on ffmpeg.js is at: https://github.com/mdhsl/ffmpeg.js

This latest build also includes H265/HEVC and VP8/VP9 codecs. Hope you'll find what you need there.

MathisTLD, 2020-06-05:
Merci !!!

closed 2020-06-08 by alexrobin