opensensorhub/osh-js #187: Image Draping broken

mdhsl, 2020-07-14:
The Image Draping is not working anymore, neither on the react or vuesjs demos. The draping is not synchronized with the map

[image]

mdhsl, 2020-07-14:
It is not a bug. It is happening because the CPU is not fast enough to decode the h264 frame as fast as it needed to synchronize with orientation and position. It is falling behind.  That kind of use case should use the new next data synchronizer algorithm to be sure the data are synchronized together between sensor in a same entity

mikebotts, 2020-07-15:
Hi Matt. Should this issue NOT be closed but maybe just redefined as something that needs to be further looked at in the future. Its still shown not to be working even if it's not a bug, right?

Dr. Mike Botts
Botts Innovative Research, Inc.
http://www.botts-inc.com <http://www.botts-inc.com/>
https://opensensorhub.org <https://opensensorhub.org/>
256-652-0165

mdhsl, 2020-07-15:
Well we can re-open the issue but I am not sure about the solution for this

alexrobin, 2020-07-15:
I would just modify the demo to use the Data Synchronizer.

mdhsl, 2020-07-15:
It will not fix the problem. The problem is the too much data are going to be decoded by the view and take some lag because the CPU cannot decode as fast as it should. So we get a lag

alexrobin, 2020-07-15:
Yeah you're right.
Why is decoding taking such a long time in this case?

mdhsl, 2020-07-15:
I think the resolution + the framerate are too much for my laptop. On my desktop, I don't reproduce the problem. You can reproduce this using the chrome dev tool which allows to slow down the CPU to 25% for example.

A solution would skip frame into the FFMpegView when we are too far from the current time but it is simple to implement

alexrobin, 2020-07-15:
It's strange because it used to be running fine on my laptop.
I haven't tested the latest version though.

mdhsl, 2020-07-16:
Try to use the CPU throttle using chrome: dev tool -> performance tab -> CPU throttling

You should reproduce the problem

mdhsl, 2021-03-14:
Tested on last release and it seems working fine even using CPU throttling 6x. I think we can close this issue

closed 2021-03-14 by mdhsl