opensensorhub/osh-js #245: Remove duplicated code from DataSource worker onmessage method

alexrobin, 2020-08-19:
https://github.com/opensensorhub/osh-js/blob/master/source/osh/datareceiver/workers/SweJson.worker.js#L10

We can create a `handleMessage()` method in `DataSourceHandler` that contain this code and then just call `dataSourceHandler.handleMessage(event)` in each datasource worker implementation

mdhsl, 2020-08-23:
This would be a good optimization, we have to pay attention to 2 things:
- https://github.com/opensensorhub/osh-js/blob/master/source/osh/datareceiver/workers/SweJson.worker.js#L13
This is mapping the worker onData method to post a message 
- https://github.com/opensensorhub/osh-js/blob/master/source/osh/datareceiver/workers/SweJson.worker.js#L21
This returns the timeStamp using the postMessage as well

One solution would to pass the worker to the handleMessage() function such like handleMessage(message, workerInstance).

Do you see other solution to handles that cases?

alexrobin, 2020-08-25:
I would pass the worker itself (i.e. self) in the constructor of `DataSourceHandler` (line 4 in each worker implementation). This way we can call `worker.postMessage()` wherever we need to in `DataSourceHandler` code.

closed 2020-08-28 by mdhsl