opensensorhub/osh-js #258: Bug in DataSynchronizer when one datasource reconnects

alexrobin, 2020-09-02:
It seems that when a data source gets disconnected and then reconnected, the data synchronizer doesn't behave properly, leading to either increased latency or loss of synchronization.

I think the data synchronizer needs to be notified of any datasource reconnection so that the associated state can be properly reset (e.g. data queue, measured latency, etc.)

mdhsl, 2020-09-04:
It also leads to other problem such as Control (Forward/Backward/Pause/Play)

alexrobin, 2020-09-04:
Yes, we need to build an showcase example with several data sources (e.g. 3 times the same video) under the same data synchronizer, and then time control the synchronizer with the playback widget to properly test this case.

closed 2020-10-02 by mdhsl