opensensorhub/osh-js #274: [datasynchronizer]  remove direct push

mdhsl, 2020-09-26:
https://github.com/opensensorhub/osh-js/blob/master/source/osh/datasynchronizer/DataSynchronizer.worker.js#L38
The datasynchronizer is currently accepting direct push w/o passing by the Broadcast channel. In that specific case:
1) it won't work because only the data is passed to the DataAlgo w/o the timestamp
2) we don't want to handle 2 cases, direct and broadcast channel.

The best option is to remove the best push and to handle only the broadcast incoming data channel.

mdhsl, 2020-10-01:
@alexrobin

alexrobin, 2020-10-02:
I think we should keep the direct push in case we combine the datasynchronizer w/ something else in the same thread/webworker at some point. We should just fix [line 36](https://github.com/opensensorhub/osh-js/blob/dde8acaef05fe0cc2c9b91632b4be41359974240/source/osh/datasynchronizer/DataSynchronizer.worker.js#L36) to include the timestamp:
`dataSynchronizerAlgo.push(event.data.dataSourceId, event.data.data);`

closed 2020-10-02 by mdhsl