opensensorhub/osh-js #334: Improve Time Control component

alexrobin, 2021-01-22:
Improve the existing `Control` Vue.js component:

- Rename it to `TimeController`
- Make it work with any `TimeSeriesDataSource` or a `DataSynchronizer`
- Add a checkbox to the right of the playback buttons to toggle between "archive replay" and "real-time" modes.
  * In real-time mode, there should be only one (unselectable) slider handler, and playback buttons should be disabled.
  * When going back to replay mode, the slider handles should be restored to the last used start and end time, and the datasource should be reset/stopped until the user moves the range slider to a new position or voluntarily click on the "play" button
- Add 2 buttons to increase/decrease the replay speed factor dynamically

And to help others use the new component:
- Add a showcase example where it controls a single data source (e.g. gps marker)
- Add a showcase example where it controls a data synchronizer (e.g. marker + video)

mdhsl, 2021-01-28:
Currently, the View emits an 'onchange' event when the slider is moved and it is up to the developer to capture this event and update the dataSource manually.

It's better to update the dataSource automatically in the View and then fire the onchange to allow further actions afterwards.

mdhsl, 2021-01-28:
Sliding the time slider using DataSynchronizer is also not working

alexrobin, 2021-02-09:
I tested the new component in dev and noticed the following issues:

- After changing the requested replay period several times and then moving to real-time, the chart sometimes doesn't redraw properly (several data points stuck at the right of the chart for a few seconds until it redraws properly). It also happens when moving from real-time to replay mode (no data points on the chart for a while). In both cases, there seems to be a bad data point that prevents the X axis to be resized properly (see [screen capture](https://drive.google.com/file/d/17f4t58lxN_b5mpAoWSINyNEbtKGfUWi7/view?usp=sharing)).

- The cursor seems to oscillate left and right while playing (playback mode)

- I don't think we should require the user to press play again after moving the cursors on the time bar. IMO pressing play should only be required when switching back from real-time mode to replay mode, but maybe it could be a config option.

- Hard to increase/decrease replay time as it generates a request every time one clicks +/- buttons. We should leave time to allow several clicks on +/- before a new request is made to the server.

- Timestamps don't update fast enough when one drags the cursor on the time bar (need to update in real-time so one can see the exact time selected)

mdhsl, 2021-02-20:
#345

closed 2021-02-20 by mdhsl