opensensorhub/osh-js #344: Polyline Support in CesiumView

nickgaray, 2021-02-18:
It seems we have lost support for polylines in CesiumView.

Here is the stop gap solution implemented for what I am working on.

CesiumView.prototype.polylines = [];

CesiumView.prototype.addPolyline = function(properties) {
    let width = properties.width;
    let entity;

    if (properties.type === 'bearing') {
        // console.log('Creating LOB Styler', this.viewer);
        let cViewer = this.viewer;
        entity = this.viewer.entities.add({
            polyline: {
                positions: new Cesium.CallbackProperty(function (time, result) {
                    return properties.locationArray;
                }, false),
                width: properties.width,
                material: new Cesium.Color.fromCssColorString(properties.color),
                clampToGround: true
            }
        });
    } else if (properties.type === 'trail') {
        entity = this.viewer.entities.add({
            polyline: {
                positions: new Cesium.CallbackProperty(function (time, result) {
                    return properties.locationArray;
                }, false),
                width: properties.width,
                material: new Cesium.Color.fromCssColorString(properties.color)
            }
        });
    }
    // console.log(this);
    let id = 'polyline-' + randomUUID();
    entity._dsid = id;
    this.polylines[id] = entity;    // [NOTE] Make sure this is accessed correctly by the rest of OSH

    return id;
};

nickgaray, 2021-02-18:
Probably helps to have a styler with it :) 

/**
 * Creates a line of bearing Styler
 */
class LineOfBearing extends Styler {

    /**
     *
     * @param properties
     */
    constructor(properties) {
        super(properties);

        this.properties = properties;
        this.origin = {lat: 0, lon: 0, alt: 0};
        this.bearing = 0;
        this.locationArray = [];
        this.color = 'red';
        this.opacity = 1.0;
        this.width = 1;
        this.length = 10;
        // this.smoothFactor = 1;

        if (hasValue(properties.color)) {
            this.color = properties.color;
        }

        if (hasValue(properties.width)) {
            this.width = properties.width;
        }

        if (hasValue(properties.opacity)) {
            this.opacity = properties.opacity;
        }

        if (this.checkFn("lobFunc")) {
            let fn = function(rec,timeStamp,options) {
                if (this.locationArray.length === 2) {
                    this.locationArray.shift();
                    this.locationArray.shift();
                }
                let location = properties.lobFunc.handler(rec,timeStamp,options);
                this.locationArray.push(...location);
            }.bind(this);
            this.addFn(properties.lobFunc.dataSourceIds,fn);
        }
    }

    /**
     *
     * @param view
     */
    init(view) {
        // console.log('Initializing Polyline Bearing')
        super.init(view);
        if (typeof (view) != "undefined") {
            view.addPolyline({
                type: 'bearing',
                locationArray: this.locationArray,
                color: this.color,
                opacity: this.opacity,
                width: this.width
            });
        }
    }

    /**
     *
     * @param dataSourceId
     * @param rec
     * @param view
     * @param options
     */
    setData(dataSourceId, rec, view, options) {
        let setData = super.setData(dataSourceId, rec, view, options);
        if (setData) {
            if (typeof (view) != "undefined" && typeof view.updatePolyline === 'function') {
                view.updatePolyline(this, rec.timestamp, options);
            }
        }
        return setData;
    }

    /**
     *
     */
    clear() {
        this.locations = [];
    }
}

mdhsl, 2021-02-20:
Map Views now inherit MapView.

The implementation of markers and polylines are now simplified thanks to the methods of the parent class. 
For the View Cesium, the Marker part has already been done but I don't remember that the View Cesium already supported Polylines.

To add this part, the best would be to follow the same principle as the other Map Views to be homogeneous in the implementation.

nickgaray, 2021-02-23:
Ok, I am implementing the Polyline support in CesiumView using the current design pattern for MapView derived classes.
The Polyline support was in Cesium but appears to have been in a forked project we used as a stop gap before.

I will also add to osh-ext->layer a new LineOfBearing layer as its behavior is a specific case of PolylineLayer.  This way I will not pollute PolylineLayer with use case specific logic.

mdhsl, 2021-02-23:
Great ! I think you can add your layer into the osh directory (no need to use osh-ext)

nickgaray, 2021-02-24:
Made necessary changes to support polyline.  Decided to keep LineOfBearingLayer seperate for now until we can test more thoroughly.

closed 2021-02-27 by mdhsl