opensensorhub/osh-js #348: Rename packages

alexrobin, 2021-02-28:
Let's rename some packages before building an NPM dist version:

dataconnector -> connector
datareceiver -> datasource
datasender -> command
datasynchronizer -> timesync

mdhsl, 2021-03-05:
Some questions here:

* Do we have to refactor DataConnector class into Connector as well?
* DataSynchronizer class into TimeSync?
* package osh -> core?
* package osh-ext -> ext ?
* package osh-vue -> vue ?

alexrobin, 2021-03-05:
No, don't rename the `DataConnector` and `DataSynchronizer` classes, just the packages.
And ok to rename the top level packages as you suggest:
- osh -> core
- osh-ext -> ext
- osh-vue -> vue

mdhsl, 2021-03-05:
Like for example OpenLayers, we have to do something like:

mdhsl, 2021-03-06:
done 29b8ebd6efef4054c73c42b9f70086d6141c779c

closed 2021-03-06 by mdhsl