opensensorhub/osh-js #349: DataSynchronizer returns true incorrectly

nickgaray, 2021-02-28:
```
   async isConnected() {   
        for(let ds of this.dataSources) {
            if(await ds.isConnected()) {
                return false;
            }
        }
        return true;
    }
```

Needs to be

```
   async isConnected() {
        let isConnected = false;
        for(let ds of this.dataSources) {
            if(await ds.isConnected()) {
                isConnected = true;
                break;
            }
        }
        return isConnected;
    }
```

mdhsl, 2021-03-06:
The fix would invalidate the state if at least one of the dataSource is not connected : 
```js
    async isConnected() {
        for(let ds of this.dataSources) {
            if(!(await ds.isConnected())) {
                return false;
            }
        }
        return true;
    }
```

I am making the fix

mdhsl, 2021-03-06:
done 071479894085c63bb3506c2307a777269457e138

closed 2021-03-06 by mdhsl