opensensorhub/osh-js #414: File DataSource non-functional due to 3rd party dependency issues?

Frikster, 2021-05-13:
I am trying to see if I can run [this example](https://github.com/opensensorhub/osh-js/blob/master/showcase/examples/video-map-multiple-datasource/video-map-multiple-datasource.js) from the showcase except sourcing my own local video file and csv with telemetry data.

I am trying to use the [File Protocol](https://opensensorhub.github.io/osh-js/latest/site/guide/protocol/general.html#file-protocol-file) and made a quick edit in the video-map-multiple-datasource.js file switching the `videoDataSource` variable to instead be sourced from:

```js
    const videoDataSource = new File("drone-Video",{
        paths: ['https://drive.google.com/file/<link to MP4>']
      });
```

This leads me to errors of the form:

 ```
 ./node_modules/stream-browserify/index.js
Module not found: Error: Can't resolve 'readable-stream...
```

And from looking at `stream-browserify` and `readable-stream` I notice that a [relatively recent issue](https://github.com/browserify/browserify/issues/1999) on browserify points out that dependencies are presently botched between the two libraries.

The preferred method of using OSH appears to be to host files through the OSH server with correct ontologies defined etc etc, a route that presumably doesn't lead to the above issue. 

Before I dive into OSH-Server docs I just want to ask whether my conclusion to my preliminary investigation is correct, whether I'm trying to do something that would actually work if not for the dependency issues, or even whether my small test is not how to use the File Protocol at all and how I can rectify my understanding.

Thank you in advance.

mdhsl, 2021-05-13:
Hi! Thanks for reporting.

Did you take a look at the full Earthquake example using File protocol? : https://github.com/opensensorhub/osh-js/tree/master/demos/earthquake

mdhsl, 2021-05-13:
You may need to include 'fs' as dependency into your package.json

mdhsl, 2021-05-13:
Thanks for reporting. There are multiple things, first of all, the File dataSource uses Loaders.gl and only some loaders are supported. You can use the FetchApi directly to fetch your video data in binary format without going through the usual parsers. 

One important thing with videos, the decoders provided in OSH are configured to decode full NAL UNIT. The OSH server slices the data into NAL UNIT and the decoder (FFMPEG or WebCodecAPI) decodes this NAL. If you want to read your video from a local file or from the network, you will have to implement in the parser a system that splits for you the received bytes into NAL UNIT and send to the decoder NAL by NAL. This was done like this in the first versions of OSH, then we assumed that the server would systematically send complete NALs. 

I don't find it illogical to develop a new parser to parse a custom stream and split it into NALs. 

Last important thing, your example suggests that you use mp4, but mp4 is a container, it can contain various types of encoded files such as H264. So inside the mp4, you will find metadata and your video stream in h264 (if it has been encoded in this format). The ffmpeg decoder does not decode mp4 but only the raw video stream, that is H264, H265, vp8, vp9 etc.. So you will have to extract your video stream from mp4 format. In the first versions of OSH, there was also a parser to extract information from the mp4 format (see tag 1.1 of the repo).

I've made a small example in a separate branch that allows you to read a binary format from a file:

https://github.com/opensensorhub/osh-js/tree/File_DataSource_non_functional_due_to_3rd_party_dependency_issues_414/showcase/examples/video-h264-file

I hope this helps you. If you want to contribute to the repo and add the datasource to parse the NAL UNIT, you are welcome :)

closed 2021-06-06 by mdhsl