opensensorhub/osh-js #422: TimeSynchronizer gets messed up and out of sync

mikebotts, 2021-06-01:
When using data from multiple sources, some or all data streams may get out of synch. This happens both within our COP and with the Showcase at "MultiVideo with Control". can often cause this problem with multiple changes to the time slider; may be particular susceptible when network is great.

NOTE: when data not synchronized, one can often see the time slider jump back and forth along time bar.

mdhsl, 2021-06-02:
Thanks for reporting. I cannot reproduce the problem using the MutiVideo With Control.

GIven this example of the showcase: http://opensensorhub.github.io/osh-js/dev/showcase/video-with-control-vuejs-synchronized.html

In latest or dev branch, it works fine. 

There was a problem before due to a graphic problem of the slider that was fixed some time ago. So you may still have the old version cached. Try to run the example in private mode to invalidate any potential cache of the old javascript code

Here a video to demonstrate the thing:

https://user-images.githubusercontent.com/6577027/120429093-1c667a00-c375-11eb-95a0-3b8554473454.mp4

You can validate the synchronization algo using this example too:

http://opensensorhub.github.io/osh-js/dev/showcase/datasources-synchronized.html

mikebotts, 2021-06-02:
Yes it's easier to get things out of synch when you alter the replay speed a few times. I intended to list that as a separate issue. It's a little harder to cause the sync issue when it running at 1x speed. With the Showcase, the time differences between data sources may remain small (1-2 sec) and show itself as a jitter of the time slider.  However, I too am having a hard  time replicating it today after doing a refresh on the page.

We have definitely noticed the issue being much more exaggerated for our COP as we deal with bringing in data from several sensors, but I cannot guarantee that it not an issue with our COP code. I'm assuming that the changes you made to the the toolkit code have been incorporated into our COP.  There are some restrictions perhaps on me showing you examples of the TE data running on our COP but let me explore that a little more. 

Let's don't close out this issue for a bit, in order to give ourselves a little more time to explore the issue.
Thanks.

mdhsl, 2021-06-02:
Ok. I know there was an issue on that and I have made a fix for that specific problem. But once you have already load a demo, the browser does not reload and use the cache instead. This is another issue I have to work on.

So the only way to be sure to load the last version without cache is to load the showcase in private mode.

@alexrobin did you try in private/incognito mode?

alexrobin, 2021-06-03:
@mdhsl yes, the problem still occurs after clearing the cache or in incognito mode. It's easy to reproduce on my computer as soon as you set replaySpeed to 2x and skip to another position in the time bar.

I'm not sure if the problem is due to the replay speed not being accounted for correctly in the synchronization algorithm or if the problem also exists at 1x speed but is just less visible.

mdhsl, 2021-06-06:
@mikebotts  @alexrobin  

I have created 1 example which shows 3 separate synchronized video streams. We can see that the algorithm of the datasynchronizer works correctly.

http://opensensorhub.github.io/osh-js/dev/showcase/multi-datasources-synchronized.html

Moreover, I made a correction in the TimeController to display a warning when a datasource is out of sync.
The problem you should have had was that some data were late and the TimeController displayed them anyway. From a certain point on, data that exceeds the timeout defined in the datasource is no longer synchronized because it is considered out of sync. This behavior is normal but was not managed correctly at the view level.

From now on, the TimeController will skip all out of sync data and display a blinking orange icon with the name of the concerned dataSource in tooltip. If the dataSource becomes synchronized again, it will be removed from the list and the icon will not be displayed anymore (unless other dataSources are out of sync of course).

An important thing to remember is the use of the **timeOut** and **bufferingTime** dataSource properties which define when data is considered to be out of sync.

Here are 2 videos demonstrating the problem before: 

https://user-images.githubusercontent.com/6577027/120934040-081bd780-c6fd-11eb-9d34-cdac83ef82e2.mp4

then after the correction:

https://user-images.githubusercontent.com/6577027/120934053-1669f380-c6fd-11eb-8234-c0e65f28187f.mp4

Can you confirm that it is ok and we can close this ticket

mdhsl, 2021-06-06:
#425

alexrobin, 2021-06-10:
After testing more, I can reproduce the problem systematically with the following steps:

- Set replay speed to 2x
- Slide timebar to skip forward: sync is ok
- Slide timebar to skip backward: the 4 data sources are out-of-sync and orange icon is blinking

If I pause and resume playback, everything gets synced correctly again

mikebotts, 2021-06-10:
Yes. Alex's procedure reproduces the problem for me as well.

mdhsl, 2021-06-11:
there were several problems:
* problem of resetting some properties in the synchronization algo (buffering + DS object)
* problem of broadcast channel buffer which continued to send data from its buffer when disconnected => added a packet versioning system to handle this kind of case
* Datasource out of sync problem that you have to skip in the TimeController (a warning out of sync icon now displays the list of out of sync datasources)

I have created a new demo to specifically test the synchronization with high frequency incoming data and using the TimeController:

http://opensensorhub.github.io/osh-js/dev/showcase/videodata-with-control-vuejs-synchronized.html

and you can retest the video example:

http://opensensorhub.github.io/osh-js/dev/showcase/video-with-control-vuejs-synchronized.html

closed 2021-06-17 by mdhsl