opensensorhub/osh-js #445: Use Any Projection with osh-js

drewbotts, 2021-07-20:
When using osh-js with a map projection that is EPSG:4326, markers and other layers are being transformed to EPSG:900913. This should be generic to any projection we would like to use.

mdhsl, 2021-07-22:
This is not a global problem with osh-js, only with the OlView

mdhsl, 2021-07-22:
But I think the best way is to project coordinates into the Style locations function because each Map has a specific way to handle projection,example: MapBoxView only works with 4326 coordinates. 
What I can do is to add the Map projection into JsDoc to know if we have to project coordinates location or not

mdhsl, 2021-07-22:
Fixed #447

alexrobin, 2021-07-23:
I would rather document that all views accept only 4326/4979 data for now, and that each view is responsible to transform to its internal projection, whenever needed.

For OpenLayers this means transforming to the projection used by the map using OL internal functions.

alexrobin, 2021-07-23:
Ok I think that's what you did in the end so all good.

closed 2021-07-23 by alexrobin