opensensorhub/osh-js #527: Allow clients to specify label styling

nickgaray, 2021-09-30:
[image]
Label styling, such as color and optionally border should be allowed for any PointMarkerLayer.

nickgaray, 2021-09-30:
The example image provided shows a black label with white outline

mdhsl, 2021-09-30:
I agree, the biggest difficulty is to have something homogeneous between the different views. But for now, we can add the properties in the Layer and implement them in the Cesium view

alexrobin, 2021-09-30:
Instead of doing something specific for Cesium label outlines, I think it would be better to have a general mechanism to add any custom configuration that is specific to a particular view implementation (i.e. not handled by generic styler properties). It could be done via a callback providing the graphical object created by the view impl for example, so it can be further customized.