opensensorhub/osh-js #563: add ability to pass divID in map properties (OpenLayerView)

drewbotts, 2021-11-18:
- May need to be included in other map views

currently having to use this work-around:
`let temp = map.getTarget();

            let olView = new OpenLayerView({
                map: map,
                layers: [pointMarker],
                autoZoomOnFirstMarker: true
            });

            map.setTarget(temp);`

drewbotts, 2021-11-18:
to clarify this is when passing in an already create openlayers map to osh-js

mdhsl, 2022-01-23:
I think it is not relevant because how you will interact with OL objects (layers/markers/features etc..) without the Map Object itself?

alexrobin, 2022-01-24:
I think what @nickgaray and @drewbotts mean is that we need to pass both the Map object and the div ID.

But I think passing the Map object should be sufficient, provided that in this case we don't assign our own div ID (i.e. don't do what's on [line 231](https://github.com/opensensorhub/osh-js/blob/master/source/core/ui/view/map/OpenLayerView.js#L231))

The owner of the Map should decide what div it is attached to, so if we pass in a Map it means the caller is responsible for attaching the Map to a div.

alexrobin, 2022-01-27:
Exactly. I even think all lines between lines 231 and 249 should move inside the `if`.
If the map is provided externally, we probably don't want to modify it except add our vector layer so we can draw into it.

closed 2022-01-27 by mdhsl