opensensorhub/osh-js #564: add ability to set label color and size in OpenLayerView

drewbotts, 2021-11-18:


mdhsl, 2022-01-23:
Could you provide more precision about your issue?
Label of marker I guess?
size of? icon? marker? polygon? linestring?

drewbotts, 2022-01-24:
Specifically labelColor and labelSize in the PointMarker layer. They don't to be used in an OpenLayerView.

mdhsl, 2022-01-28:
Note: After investigation OL does not support to specify the size of the icon.The property size of the icon does not allow to change the size of the icon itself but The property size of the icon does not allow to change the size of the icon itself but the sub-rectangle to be used from the original icon image (sprite) as specified here (https://openlayers.org/en/latest/apidoc/module-ol_style_Icon-Icon.html).

OL uses the default icon size; the only way to change the size is to use the scale property which is the ratio of the iconSize property / original size.

The problem is that we know the original size of the icon only after adding it to the DOM, so we see the icon appear at its original size before the scale. So this is not appropriate.

To sum up, OL does not support specifying the size of the icon. You have to use the iconScale property of the PointMarkerLayer and this will work

mdhsl, 2022-01-28:
I will now work on labelSize & labelColor

mdhsl, 2022-01-28:
The same comment can be applied to labelSize. I have added a new property to PointMarkerLayer called 'labelScale'. Here the result for the OLVIew:

[image]

mdhsl, 2022-01-28:
done

closed 2022-01-28 by mdhsl