opensensorhub/osh-js #691: DataSynchronizer support for dynamically adding and removing datasources

nickgaray, 2022-11-27:
Use case:

- User has a number of 'visualizations' (composed of one or more data streams)
- User has ability to switch from real-time to playback
- User switches to playback mode (DataSyncrhonizer is created without any datasources added)
- User selects which visualizations to view/review (adding corresponding datasources to synchronizer)
- User connects data synchronizer

Extension to use case:
- User may elect to stop viewing a visualization (removing corresponding datasources from synchronizer)

Issue:
Adding a datasources to synchronizer before connecting, the async setDataSynchronizer(dataSynchronizer) is not being invoked and when the request is made it is made without a time range such as this:
 
`ws://localhost:8282/sensorhub/api/datastreams/r5adq419u4h4c/observations?format=application/om%2Bjson`

mdhsl, 2022-11-30:
What is not supported today is adding a dataSource after the dataSynchronizer is started.

nickgaray, 2022-11-30:
The other operation is to remove a datasource, this is also currently not supported

`dataSyncrhonizer.removeDataSource(someDataSource)`

So, I agree with the adding but it would be great to also remove, in addition to the dataSyncrhonizer not reporting it is connected if it does not have any dataSources or if all its data sources are disconnected.

mdhsl, 2022-12-05:
Example here: https://opensensorhub.github.io/osh-js/dev/showcase-dev/datasynchronizer-dynamic-add-remove.html

closed 2022-12-05 by mdhsl